orbcode/orbuculum #3: make Error

odunboye, 2017-07-13:
typing make after cloning gives the following error

/bin/sh: ofiles/git_version_info.h: No such file or directory
make: *** [get_version] Error 1

mac osx sierra
kindly help.

mubes, 2017-08-12:
Sorry, didn't see this :-(

So, the problem is that a git version stamp is generated automatically by the makefile. If you do not have git installed and running then that will cause you issues.  You can put a file into ofiles/git_version_info.h containing something like the following (and remove the git_hash_to_c.sh reference in the makefile).
```
#define GIT_DIRTY         1
#define GIT_HASH          0x3bae84eb
#define GIT_BRANCH        "master"
#define BUILD_DATE        "2017-07-28 23:03:00+0100"
```
Frankly, the preferred option is to pull the build across as a git tree and build that, and then the problem should go away, and you stay current with the git changes too.

Regards

DAVE

karlp, 2017-08-29:
nope. it's not quite that.  you get that error even in a git clone.
```
karlp@teros:~/src/orbuculum (master)$ make
/bin/sh: ofiles/git_version_info.h: No such file or directory
Makefile:157: recipe for target 'get_version' failed
make: *** [get_version] Error 1
karlp@teros:~/src/orbuculum (master)$ git log --oneline -n 1
50e669c Changes to orbtrace to use memory elements
karlp@teros:~/src/orbuculum (master)$
```

karlp, 2017-08-29:
(I think you just need a mkdir -p call)
```
diff --git a/Makefile b/Makefile
index bc56fa6..1e742c2 100644
--- a/Makefile
+++ b/Makefile
@@ -154,6 +154,7 @@ CFILES += $(App_DIR)/itmDecoder.c $(App_DIR)/tpiuDecoder.c $(App_DIR)/generics.c
 all : build 
 
 get_version:
+       $(Q)mkdir -p $(OLOC)
        $(Q)$(GET_GIT_HASH) > $(OLOC)/$(GIT_HASH_FILENAME)
 
 $(OLOC)/%.o : %.c

```

mubes, 2017-10-06:
Agreed. Thanks for the fix.

closed 2017-10-06 by mubes

odunboye, 2018-01-06:
After running make

_ld: library not found for -liberty
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [orbuculum] Error 1_

Mac OS X high Sierra

Kindly help

mubes, 2018-01-06:
Hi Karl,

Try removing that library from the linker configuration ... It was added
because SuSE needed it, but it may not be required in your configuration -
let us know how you get on and I'll put in an automatic variation if it
fixes the issue.

Regards

Dave

odunboye, 2018-01-06:
Hello Dave,

Thanks for this nice software.
I removed the '-Liberty' and ran make again then got  below

_Completed build of orbuculum
Completed build of orbcat
 Compiling Src/orbtop.c
Src/orbtop.c:327:94: warning: format specifies type 'long' but the argument has type 'uint64_t' (aka 'unsigned long long') [-Wformat]
                    fprintf( stdout, "%3d.%02d%% %8ld ", percentage / 100, percentage % 100, report[n].count );
                                                 ~~~~                                        ^~~~~~~~~~~~~~~
                                                 %8llu
Src/orbtop.c:367:93: warning: format specifies type 'long' but the argument has type 'uint64_t' (aka 'unsigned long long') [-Wformat]
        fprintf( stdout, "%3d.%02d%% %8ld Samples" EOL, totPercent / 100, totPercent % 100, samples );
                                     ~~~~                                                   ^~~~~~~
                                     %8llu
Src/orbtop.c:371:100: warning: format specifies type 'long' but the argument has type 'uint64_t' (aka 'unsigned long long') [-Wformat]
        fprintf( stdout, "%3d.%02d%% %8ld of %ld Samples" EOL, totPercent / 100, totPercent % 100, dispSamples, samples );
                                     ~~~~                                                          ^~~~~~~~~~~
                                     %8llu
Src/orbtop.c:371:113: warning: format specifies type 'long' but the argument has type 'uint64_t' (aka 'unsigned long long') [-Wformat]
        fprintf( stdout, "%3d.%02d%% %8ld of %ld Samples" EOL, totPercent / 100, totPercent % 100, dispSamples, samples );
                                             ~~~                                                                ^~~~~~~
                                             %llu
4 warnings generated.
 Compiling Src/symbols.c
Src/symbols.c:204:26: error: no member named 'st_mtim' in 'struct stat'
            ( memcmp( &n.st_mtim, &( ( *s )->st.st_mtim ), sizeof( struct timespec ) ) ) ||
                       ~ ^
Src/symbols.c:204:49: error: no member named 'st_mtim' in 'struct stat'
            ( memcmp( &n.st_mtim, &( ( *s )->st.st_mtim ), sizeof( struct timespec ) ) ) ||
                                     ~~~~~~~~~~ ^
Src/symbols.c:205:26: error: no member named 'st_ctim' in 'struct stat'
            ( memcmp( &n.st_ctim, &( ( *s )->st.st_ctim ), sizeof( struct timespec ) ) )
                       ~ ^
Src/symbols.c:205:49: error: no member named 'st_ctim' in 'struct stat'
            ( memcmp( &n.st_ctim, &( ( *s )->st.st_ctim ), sizeof( struct timespec ) ) )
                                     ~~~~~~~~~~ ^
4 errors generated.
make: *** [ofiles/Src/symbols.o] Error 1_

What could be the problem?
Mac OS X high Sierra

mubes, 2018-01-06:
Hi,

Ive not got access to a pc at the moment but it seems that the fields
st_ctim and st_mtim appear to be called st_mtime and st_ctime  respectively
on osx. They might also be called st_mtimespec and st_ctimespec if it's
being compiled as POSIX (I don't think it will be). Correct the field names
on the errant lines and see how you go.  The warnings we can easily fix
later but if this fixes the errors then at least you can get going.

Regards

Dave

Src/orbtop.c:371💯 warning: format specifies type 'long' but the argument

pvonmoradi, 2020-12-16:
> After running make
> 
> _ld: library not found for -liberty clang: error: linker command failed with exit code 1 (use -v to see invocation) make: *** [orbuculum] Error 1_
> 
> Mac OS X high Sierra
> 
> Kindly help

In Xubuntu 18.04 I installed these to build the app:
`sudo apt install binutils-dev libelf-dev libiberty-dev`
(note the last one for -liberty)

karlp, 2020-12-17:
please make a new issue, your issue is completely unrelated to this make directory path error from two years ago