orbcode/orbuculum #4: file input runs and exits before clients can connect

karlp, 2017-08-29:
I'm trying to use the file input mode with the output file written by openocd.  (So that I can get the SWO data from say, stlink instead of having to have an extra usb-uart)  but a) it doesn't follow the file, it just reads it once, and b) it simply reads the file, passes it to "all"  clients and exits.  It does this... very quickly.  Far far far faster than one could possibly run any of the other orb tools to actually connect and _be_ one of these clients :)

karlp, 2017-09-07:
I tried this again recently by trying to configure openocd to write the swo trace to a fifo instead of a file, and telling orbuculum to read from teh fifo.  It now at least "runs" but it doesn't seem to get any data from it, it just sits there.  I know openocd's writing to the fifo, because I can manually cat it and get the swo trace I expect.

karlp, 2017-09-12:
Solution here is to just use netcat with openocd's captured file.  In this case, using plain STLink hardware.

```
# openocd configured to write to swo.dump.log via...
tpiu config internal swo.dump.log uart off <corespeedinhz>

$ tail -f swo.dump.log | nc -v -v -l 9999 -k
```

And orbuculum in another window reading that.  It's not really "segger" format, it's just "raw swo pin data"

```
./ofiles/orbuculum -g 9999 -b md/ -c 0,text,"%c"
```

Perhaps this can just be addressed as documentation, rather than trying to make it follow the file itself.

mubes, 2017-10-06:
Originally the file input mode was really just there for development purposes. It doesn't work well because of the timing relations that are implicit in the tpiu decoder....and, as you've found, it doesn't wait for data to arrive in the file. I'll leave this open for now and look at enhancing the file handling to wait for data so it can be used with tools like openocd.

Sorry this has taken a while to respond to....been away creating the parallel trace fpga configuration, but now is time to solidify the core apps.

karlp, 2017-10-06:
no problem, glad to see you back again :)  the tail -f | nc is working very well anyway, just wasn't going to make a bunch of PRs for documentation on that sort of thing.

mubes, 2017-10-08:
If you look in the Devel branch I added the file tracking capability but it's not extensively tested. It now won't drop out at the end of the file but rather will continue to wait for new input (CTRL-C to interrupt). Please give it a try and see if it avoids you needing the nc hack. There are some other not well tested changes in that branch too so ymmv but I think it should be a reasonably safe experiment.

karlp, 2017-10-09:
Hrm, I don't really see anything in the Devel branch that adds this?  It's a bit hard to see from your commits though, you've only got a single commit there which fixes mkdir, changes transfer sizes for everyone and then a whole pile of seemingly irrelevant EOL printf changes all in the same commit.  Are you missing a git push from your end perhaps?

mubes, 2017-10-09:
User to dumb to use git. Replace user and press any key.

I'll commit the change tomorrow when I'm back in front of the machine that
has the development code on it.

Dave

mubes, 2017-10-10:
OK. Please give that a try on the Devel branch. If it works for you I'll close off this PR.

Dave

karlp, 2017-10-10:
Yep, this works, as long as I still have my "force itm sync" hack.    but that's a different issue, so this can be closed thanks!

closed 2017-10-10 by karlp