orbcode/orbuculum #9: CP2102 details

sorki, 2018-01-02:
Hi, nice work you're doing! I'm trying to run this with CP2102 but no luck so far

my orbuculum command is

`orbuculum -b swo/ -v 3 -a /dev/ttyUSB0 -s 115200`

for gdb

`prepareSWD 72000000 115200 1 0`

followed by

https://github.com/mubes/orbuculum/blob/master/Support/gdbinits/gdbinit-bmp#L35

The result is I can see the data on scope and when using screen but no output from `orbcat`.

Also tried with 921600 baudrate (both in gdb and for orbuculum, also with TPIU on/off).

I'm also going to try with BluePill and F4 discovery flashed with BMP, managed to flash HX8K board as well but no data from parallel trace yet. Will report soon.

mubes, 2018-01-02:
Hi sorki, good to have you aboard. The more people who start using this then the more solid we can make it all.  Your command line isn't quite right, try;

`orbuculum -b swo/ -v 3 -a 115200 -p /dev/ttyUSB0`

...and see if that helps you move forward.  You've also switched in the TPIU with your `prepareSWD` statement, so you might want to try this in the first instance (since you don't have the `-t` option on the orbuculum command line);

`prepareSWD 72000000 115200 0 0`

Given that you're outputting sample Program Counter data with the .gdbinit you attached you will _definitely_ hit overflow issues. Once you've got this running I'd _love_ you to try the HX8 board but only 1 or 2 bit TRACE width is working at the moment....I'm currently working on the 4 bit support. That stuff remains _very_ experimental for now, but there are very good things coming on that side.

Regards

DAVE

sorki, 2018-01-02:
Wow, it works with the command line you provided - kind of at least as I'm getting only

`80 IDLE --> TS(1) : 79 TS --> IDLE(1)`

and no samples in `orbtop`. I'm mostly interested in `orbtop` as it feels like the last missing piece in my tool-chain for quickly discovering timing / performance issues.

I'm running the latest version, seems easily package-able for Nixos. https://github.com/sorki/nixpkgs/commit/7bbe749f741fdf3ff45e987b44ee040f1aa46cac

For 1 or 2 bit tracing with HX8 do I need to change anything in toplevel verilog file?

mubes, 2018-01-02:
OK, You're collecting timestamps, so that's promising.  Here's my dwt configuration that's creating orbtop output for me right now;

```
dwtSamplePC 1
dwtSyncTAP 3
dwtPostTAP 1
dwtPostInit 1
dwtPostReset 15
dwtCycEna 1
```

For two bit output from the HX8 set .width(2) in toplevel.v and make sure you've got `prepareTRACE 2` in your gdbinit.

Regards
DAVE

mubes, 2018-01-16:
Sorki,

I've heard nothing back on this for two weeks now, so I am assuming your issues are resolved and closing the issue. Please feel free to follow up if you need more.

Regards

DAVE

closed 2018-01-16 by mubes