orbcode/orbuculum #11: bumpy , daplink ?

captcha1, 2018-02-04:
Assume stuff !

Will this work with bumpy : https://github.com/electronut/ElectronutLabs-Bumpy ?

Will it work with any DAPlink probes ? The firmware ( https://github.com/ARMmbed/DAPLink ) seems to have SWO (UART / NRZ only), but it doesn't seem to be enabled for any hardware.  I have a FRDM-K64F board (w/ K20DX chip) that runs DAPlink, so I could test with that.

The IBDAP hardware seems to have SWO : https://www.adafruit.com/product/2764 , https://www.hackster.io/armstart/ibdap-affordable-cmsis-dap-jtag-swd-debug-probe-e6d9a4 ... but can't find the firmware source.

Dunno if these have SWO : https://l-tek.si/web-shop/cmsis-dap-debug-probe/ , https://os.mbed.com/platforms/SWDAP-LPC11U35/

Thanks !

mubes, 2018-02-04:
> ElectronutLabs-Bumpy ?
> product/2764 , https://www.hackster.io/armstart/ibdap-affordable-
> cmsis-dap-jtag-swd-debug-probe-e6d9a4 ... but can't find the firmware

Hi,

From what I can see this lot will work fine on bumpy because it brings out
B7 to pin 3 of the connector. You will need a version of the bmp firmware
which is async enabled and set up for the right pinout.

The others I will look at later.. Bit busy at the moment but hopefully get
time tonight (GMT). If you're intending building something specially I
would recommend you hold off for a couple of weeks.

Dave

mubes, 2018-02-21:
Apologies for the delay in replying to this. It would appear that CMSIS-DAP optionally supports a UART presentation for SWO. If you point orbuculum at that UART then it should work perfectly... Even the gdb configuration stuff should work correctly.  Please give it a go and let us know how you get on. 

Dave

mubes, 2019-01-22:
Closed following no feedback.

closed 2019-01-22 by mubes