orbcode/orbuculum #14: orbtrace for icestick currently broken

karlp, 2018-02-21:
Something went missing?

ICE40HX1K_STICK_EVN.pcf:21: fatal error: no port `rxInd_led' in top-level module `topLevel'

6ef8cfca48a12208e7f96c4e7504112cde7315e5 removed the rxInd_led, and added a heartbeat_led?

karlp, 2018-02-21:
It also uncommented the setting of the "yellow" led, which also doesn't exist, after fixing the rxInd_led

mubes, 2018-02-21:
Karl,

The drains are well up on all of the orbtrace stuff at the moment ... take a look in the blackorb branch to see considerably more progress (but focused on the HX8 board right now). Intention is to provide the option for an integrated gdb back end that will make configuring the trace capabilities a heck of a lot easier.

DAVE

karlp, 2018-02-21:
no problem, I'm not actively looking at the trace anyway, I have an icestick, but none of my targets have the ETM pins exposed anyway.  I was just kicking the wheels, thought I might report it in case you hadn't noticed.

mubes, 2018-02-21:
No worries. I'll close this to stop it flapping in the wind...it's actively being addressed (when I get the time!) in the blackorb branch.
DAVE

closed 2018-02-21 by mubes