orbcode/orbuculum #15: Use O_NONBLOCK when opening serial port

Syncena, 2018-04-10:
Hi,

I've been playing around with Orbuculum on MacOS over the last week or two. I have a couple of tweaks in the pipeline to make it play nice on this platform but the main one is the addition of the O_NONBLOCK flag when opening a serial port.

The open(2) system call can block waiting for "carrier" with some USB CDC devices. Since there is no carrier (they're not modems) you need to open them with O_NONBLOCK, set CLOCAL (which orbuculum already does) then clear O_NONBLOCK using the F_GETFL/F_SETFL commands with fcntl(2).

The attached diff contains the fix and also format string fixes for uint64_t.

Steve
[orbuculum.txt](https://github.com/orbcode/orbuculum/files/1894135/orbuculum.txt)

Syncena, 2018-04-12:
Hi Dave,

Every day's a school day - I was unaware of the /dev/cu.* alias for USB CDC devices on MacOS.

Steve

mubes, 2019-08-04:
Hi Steve,

Work rather got the better of me, so this got badly delayed, however I've now folded in the opening changes for OSX.  The long long stuff I haven't incorporated - I get length matching warnings from the compiler when I do so, so I'm figuring that, at least on my system, its promoting everything to 64 bits already....which makes the uint32_t rather superfluous.  Exactly what kind of machine are you running on and what compilation errors do you see?

Syncena, 2019-08-05:
Hi Dave,

Wow, time flies. I too have been busy with other stuff this last year so my own project, for which Orbuculum may have been useful, was also put on the back burner.

I can't promise anything but maybe in the next few months I'll have a chance to revisit the MacOS build issues.

Steve

mubes, 2019-08-05:
Hi Steve,

No worries. I'll see if I can dust off a Mac somewhere and try a
compilation too. If there _are_ issues then it would be good to get to the
bottom of them. I'll leave this open for now until one of us has had a
chance to play with it.

Regards

Dave

mubes, 2020-09-10:
Others have this working on Mac now so I'm closing this one off.  Any problems, please ping.

closed 2020-09-10 by mubes