orbcode/orbuculum #23: Missing Libiberty on Mac OS X

hellgheast, 2019-08-05:
I would like to know if there's any alternative to libiberty on Mac OS X as i'm unable to compile orbuculum since clang and macport gcc-7 don't have the libiberty library.

mubes, 2019-08-05:
Hi there,

In general you'll often find this library bundled with binutils-dev...so
please have a look there and report back. I'd be surprised if it isn't
available _somewhere_ inside gcc7.  I'll check to see if it's replaceable
but iirc we use the demangler from it which is, I'm afraid, a pretty
fundamental part of orbtop.

I'll also check how others have got it compiling on osx. Back in the day
that's how the suite first came into being, so it should be possible.

Regards

Dave

UweBonnes, 2019-08-13:
As posted on gitter, on OpenSuse <demangle.h> is the right include:
> locate demangle.h
...
/usr/include/demangle.h
> rpm -qf /usr/include/demangle.h
binutils-devel-2.31-lp150.5.3.1.x86_64

mubes, 2019-08-13:
OK, thanks Uwe. Fix pushed for the OpenSuse case...perhaps that also fixes the issue for OSX. Waiting for feedback but will close this in a week if I don't hear further.

hellgheast, 2019-08-15:
Hello,
I tried to compile binutils and other libs that should have libiberty but it doesn't seem to work.
to resume at the moment there's no known fix.
One way would be to compile gcc-7 from scratch,but i didn't complete the compilation at the moment.

hardkrash, 2020-05-29:
Would the LLVM version work?
https://llvm.org/doxygen/Demangle_8h_source.html

akohlsmith, 2020-07-31:
I have this working in OSX by using binutils from homebrew, building from source and editing the recipe to include `--enable-install-libiberty` to the configure options. I then added the homebrew install paths to LDFLAGS and INCLUDE_PATHS in the Makefile.

The last bit, elf.h, I succeeded by changing `#include <elf.h>` to `#include "elf.h"` and putting the elf.h from the arm-none-eabi toolchain into the Orbuculum Src directory.

Gasman2014, 2020-07-31:
Thanks @akohlsmith that has helped. However, i can still only build orbuculum on macOS if I comment out the following (1555).

        // #ifdef OSX
        //         /* Remove the O_NONBLOCK flag now the port is open (OSX Only) */

        //         if ( ( flags = fcntl( f, F_GETFL, NULL ) ) < 0 )
        //         {
        //             genericsExit( -3, "F_GETFL failed" EOL );
        //         }

        //         flags &= ~O_NONBLOCK;

        //         if ( ( flags = fcntl( f, F_SETFL, flags ) ) < 0 )
        //         {
        //             genericsExit( -3, "F_SETFL failed" EOL );
        //         }

        // #endif

orbit still won't build as it complains about missing demangle.h which still doesn't appear to have been installed.

akohlsmith, 2020-07-31:
Oh yes, just add `int flags`, I forgot about that. I have a PR I need to create which fixes the build on OSX.

mubes, 2020-09-10:
@akohlsmith Andrew, do you have an patch for this so we can close it off?

mubes, 2020-11-05:
As at 9f6a8e9ad68bd0594e851c68dc9f98631358c2d2 OSX building is folded back into main. Thanks a million for this @akohlsmith

closed 2020-11-05 by mubes