orbcode/orbuculum #25: orbtop segfaults if elf file disappears while running

karlp, 2019-08-14:
I'll try and find and fix this myself if I run into it again, but filing it so it doesn't get forgotten.

It's not terrible behaviour, it's just not a very graceful exit :)  (I'm presuming exiting is about the only sane thing to do.  It could stay there saying "unable to open file" forever, but not sure that's better)

mubes, 2019-08-14:
Hi Karl,

You're right, it's not very graceful. Now Orbuculum itself it nicely stable
under pretty much all reset conditions this problem becomes more obvious.

It generally occurs when a new elf is being written during a new
compilation run. The only real option I can see is to internally re-start
orbtop and re-read the elf...the big problem there is to be certain that
the elf has been fully written before opening it again.

Personally I think having these things dealt with automatically by the code
makes tools _much_ nicer to use, so I will have a dig at it.

Regards

Dave

karlp, 2019-08-14:
I think in my case it was really that the elf was _gone_ and it wasn't coming back.  (my build process generates elfs with "git describe" portions in the output, so if I commit and rebuild, I get a new elf file name)  but sure, you're very welcome to do more graceful things here :)

mubes, 2019-08-14:
Karl,
This is now fixed in the latest commit.  If the elf disappears and is re-written then it will be automatically reloaded.  If it disappears for good then it'll have a couple of attempts at reloading and then bale out rather more elegantly than it did before.

So, I'm afraid it won't work for the case where the output filename changes (although you could always ln -s it to a 'fixed' filename I guess) but it will at least be more elegant.  Turns out I'd already coded for that case, just forgot how to handle pointers :-/

Regards
DAVE

closed 2019-08-14 by mubes