orbcode/orbuculum #30: orbuculum writes weird files when ITM trace is heavily loaded

PetteriAimonen, 2020-04-29:
I had too much data being written to ITM software trace ports, which resulted in `ITM Overflow` messages as expected. However what I didn't expect is that orbuculum wrote a bunch of weirdly named files to the working directory:

<pre>
petteri@oddish:tmp$ ls -l
total 116
-rw-r--r-- 1 petteri petteri 110254 Apr 29 10:55 swo_log.bin

petteri@oddish:tmp$ orbuculum -f swo_log.bin
ITM Overflow (1)
ITM Overflow (2)
ITM Overflow (3)
ITM Overflow (4)
....
Request for write on descriptor 1 while file closed
Request for write on descriptor 6 while file closed
Request for write on descriptor 6 while file closed
Request for write on descriptor 6 while file closed
Request for write on descriptor 3 while file closed
Attempt to write to descriptor 7 while open writing 4
...

petteri@oddish:tmp$ ls -l
total 360
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 '~'
-rw-r--r-- 1 petteri petteri      2 Apr 29 10:56 '~}'$'\367\272'
-rw-r--r-- 1 petteri petteri      2 Apr 29 10:56 '~'$'\256'
-rw-r--r-- 1 petteri petteri      3 Apr 29 10:56 '<'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 '>'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\343'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\376'
-rw-r--r-- 1 petteri petteri      5 Apr 29 10:56 ''$'\352'
-rw-r--r-- 1 petteri petteri     18 Apr 29 10:56 ''$'\377'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\276'
-rw-r--r-- 1 petteri petteri      2 Apr 29 10:56 ''$'\326'
-rw-r--r-- 1 petteri petteri     17 Apr 29 10:56 ''$'\372'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\254'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\354'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\374'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\367'
-rw-r--r-- 1 petteri petteri     10 Apr 29 10:56 ''$'\303'
-rw-r--r-- 1 petteri petteri      3 Apr 29 10:56 ''$'\274'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\356'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\256'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\254\367'
-rw-r--r-- 1 petteri petteri      9 Apr 29 10:56 ''$'\372\367\272'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56 ''$'\327\017\356'
-rw-r--r-- 1 petteri petteri      0 Apr 29 10:56  4
-rw-r--r-- 1 petteri petteri      2 Apr 29 10:56  l
-rw-r--r-- 1 petteri petteri 110254 Apr 29 10:55  swo_log.bin
</pre>

I think the file should be properly synchronized etc., so I'm wondering what causes the weird files.
I've attached the swo_log.bin. I've been using git revision 1fda426.
[swo_log.bin.gz](https://github.com/orbcode/orbuculum/files/4550406/swo_log.bin.gz)

PetteriAimonen, 2020-04-29:
Hmm, actually that's probably totally corrupted data as I had the baudrate wrong. Still, possibly it should be handled better?

mubes, 2020-04-29:
Certainly shouldn't be happening.  Can you please let me know what 
command line you used, and I'll take a look.

PetteriAimonen, 2020-04-29:
`orbuculum -f swo_log.bin`

mubes, 2020-04-29:
Sorry...that was in your original report :-/  Could you please send me the swo_log.bin file that's causing the troubles? You can send it via email if you prefer.

PetteriAimonen, 2020-04-29:
Hmm, I tried to attach that to the original report also, does the link not work? https://github.com/orbcode/orbuculum/files/4550406/swo_log.bin.gz

mubes, 2020-04-29:
I couldn't see it, but I've got it now. I can see it in the original report now too. I think I need more coffee.

mubes, 2020-04-29:
Interesting way to spend my lunchbreak.  Problem was that, by default, the filewriter code is switched on so when it sees junk it can end up writing files. I've now switched that _off_ by default (because you'd know you wanted it, if you did). At the same time I also added a bit of code to prevent directory breakout in the filewriter. Please try aa01bed52acf281797cd7d203060d47e2aeb7551 and lets see if that fixes your issue.

PetteriAimonen, 2020-04-29:
Yeah, I can confirm it is fixed now! Thanks!

closed 2020-04-29 by PetteriAimonen