orbcode/orbuculum #31: Build error

Igorbunow, 2020-05-02:
Unable to build:

```
make
 Compiling Src/itmDecoder.c
 Compiling Src/tpiuDecoder.c
 Compiling Src/generics.c
 Compiling Src/orbuculum.c
 Compiling Src/filewriter.c
 Compiling Src/ftdispi.c
Completed build of orbuculum
Completed build of orbcat
 Compiling Src/orbtop.c
 Compiling Src/symbols.c
Src/symbols.c: In function ‘SymbolLookup’:
Src/symbols.c:138:35: warning: implicit declaration of function ‘bfd_get_section_vma’; did you mean ‘bfd_set_section_vma’? [-Wimplicit-function-declaration]
  138 |     uint32_t workingAddr = addr - bfd_get_section_vma( s->abfd, s->sect );
      |                                   ^~~~~~~~~~~~~~~~~~~
      |                                   bfd_set_section_vma
Src/symbols.c:140:44: warning: passing argument 1 of ‘bfd_section_size’ from incompatible pointer type [-Wincompatible-pointer-types]
  140 |     if ( workingAddr <= bfd_section_size( s->abfd, s->sect ) )
      |                                           ~^~~~~~
      |                                            |
      |                                            bfd * {aka struct bfd *}
In file included from Inc/bfd_wrapper.h:44,
                 from Inc/symbols.h:38,
                 from Src/symbols.c:50:
/usr/include/bfd.h:1206:35: note: expected ‘const asection *’ {aka ‘const struct bfd_section *’} but argument is of type ‘bfd *’ {aka ‘struct bfd *’}
 1206 | bfd_section_size (const asection *sec)
      |                   ~~~~~~~~~~~~~~~~^~~
Src/symbols.c:140:25: error: too many arguments to function ‘bfd_section_size’
  140 |     if ( workingAddr <= bfd_section_size( s->abfd, s->sect ) )
      |                         ^~~~~~~~~~~~~~~~
In file included from Inc/bfd_wrapper.h:44,
                 from Inc/symbols.h:38,
                 from Src/symbols.c:50:
/usr/include/bfd.h:1206:1: note: declared here
 1206 | bfd_section_size (const asection *sec)
      | ^~~~~~~~~~~~~~~~
make: *** [Makefile:163: ofiles/Src/symbols.o] Ошибка 1
```

```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 9.3.0-11' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
gcc version 9.3.0 (Debian 9.3.0-11)
```

```
clang -v
clang version 9.0.1-12 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/8
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/9
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7.5.0
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/8
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/9
Selected GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/9
Candidate multilib: .;@m64
Selected multilib: .;@m64
Found CUDA installation: /usr/local/cuda-10.1, version 10.1

```

mubes, 2020-05-04:
Please try the fix at https://github.com/orbcode/orbuculum/tree/get_section_fix  This is due to updated API in binutils 2.34.

mubes, 2020-05-14:
Closed after 10 days, assumed fix due to no feedback.

closed 2020-05-14 by mubes