orbcode/orbuculum #34: Develop:orbtop hard attempts to set serial config

karlp, 2020-09-10:
ca5d22f0 introduces a hard explicit call to setSerialConfig on the socket fd. I don't even know what it's trying to do.  orbtop doesn't need a serial port, and it sure doesn't seem relevant to the commit comment for ca5d22f0?

It kinda looks like you bled some attempted functionality at running orbtop directly against a dump file, or a trace spew file, instead of via orbuculum, and broke "normal" usage?  That would seem to be going against the whole principle of composing things aroudn orbuculum, and means you're going to be fixing the same sorts of things in every one of them.  Are you sure that's what you're trying to do?  Shouldn't orbtop just connect to the orbuculum service, and orbuculum takes care of how the data arrives?

Direction aside, this call completely fails on a normal socket, and exits.

karlp, 2020-09-10:
(for instance, I see you added support for this direct file support to orbcat as well, but you _didn't_ add any serial support there, which is exactly the sort of inconsistency you'll end up with you have this functionality copied into each of the tools)

mubes, 2020-09-10:
orbtop has always worked with serial devices and is a shortcut to avoid needing to run orbuculum.  The setSerialConfig is needed to accomodate high speed devices (e.g. on Linux).  You're right though, it's messy and inconsistent.  I need to think about the best solution to this but running orbuculum from an app if it isn't already running seems like a reasonable option, but it makes command line options untidy. Suggestions welcome...

mubes, 2020-09-10:
Serial options removed from orbtop...using serial should be done via orbuculum.