orbcode/orbuculum #40: Segmentation fault when using DWT to watch data value

Benjoyo, 2020-11-02:
Hey, first, thanks so much for this awesome tool!

My only issue so far is that I run into a segmentation fault when I try to emit DWT events through the ITM.

My very simple test code on my nRF52840 Preview DK looks like that:

```
int32_t test = 0x1336;

void _DWT_enable() {
  CoreDebug->DEMCR |= CoreDebug_DEMCR_TRCENA_Msk;
  ITM->LAR = 0xc5acce55;
  DWT->COMP2 = &test;
  DWT->FUNCTION2 |= (DWT_FUNCTION_EMITRANGE_Msk | (DWT_FUNCTION_FUNCTION_Msk & 0b0011));
}

int main() {

  _DWT_enable();

  while (true) {
 
    test++;
         
  }
}
```

When I run Orbuculum it will receive a few bytes and then pretty much immediately segfault:

```
$ sudo ./orbuculum -s localhost:2332 -b swo/ -P -c 0,zero,"0x%08x\n" -v 3
Orbuculum V1.10 (Git 10BC7ACB Dirty, Built 2020-11-01 13:16:26+0000)
BasePath   : swo/
ForceSync  : true
Permafile  : true
SEGGER H&P : localhost:2332
Channels   :
         00 [zero] [zero]
         HW [Predefined] [hwevent]
Established Segger Link
RXED Packet of 144 bytes
Segmentation fault
```
Sometimes it manages to write a few lines into hwevent, sometimes not. If it does, it looks something like that:

```
6,2556,0,0x0002
6,46,0,0x0002
```
 
I didn't find any documentation on what this really means (is there any?) but according to the code that is a "offset write event" (is that correct? where would I find the data value and address? Or is this event not related at all?).

Running Orbuculum without any paramenters like: `sudo ./orbuculum -s localhost:2332` segfaults as well.

Do you have an idea what causes the segfault?

mubes, 2020-11-02:
Oh dear, that doesn't look good. I think I've got a 840 DK around here 
somewhere. Leave it with me and I'll have a dig at it. It's likely to be 
something simple.

Can you perhaps grab a trace into a file and mail it to me? Netcatting 
localhost:2332 to a file for 5-10 seconds would be plenty - doesn't need 
to be long. Then I can then play that into the toolchain and see where 
it's falling over. There are so many ways you can use the tools, with so 
many different targets, that sometimes the only time we find a hiccup is 
when someone bumps into it...the trick is just to get it resolved as 
quickly as possible.

DAVE

Benjoyo, 2020-11-03:
Thanks for your reply. Does that work? Btw the example code above is not entirely correct, I have a nrf_delay_ms(1000); inside the loop.

[trace.txt](https://github.com/orbcode/orbuculum/files/5480408/trace.txt)

mubes, 2020-11-03:
Does this segfault for you when used as `orbuculum -f trace.txt` ? It 
doesn't fall over here....

Regards

DAVE

Benjoyo, 2020-11-03:
Yes, it does. I tried several traces that I collected, all segfault. Some give an ITM Overflow warning before, some don't. Can this be related to my machine? I am running this on a Raspberry Pi 3 B+. Normal ITM SendChar etc. works totally fine though.

mubes, 2020-11-03:
I guess it could be. I have one of the new Pi 400's arriving today. Give 
me some time and I'll be back. There may be endian issues. ITM Overflow 
isn't unusual when you're running over SWO, so I wouldn't worry too much 
about that (especially if you've got timestamping switched on).

mubes, 2020-11-03:
Just tried it over lunch on Ubuntu on Pi. Still works perfectly :-/  Do you get any warnings on build?  

Next step is to run it under gdb with your existing command line like; gdb --args <command line> so we can see where _exactly_ it's falling over.

Benjoyo, 2020-11-03:
Good idea, this is what gdb gave me:

```
(gdb) run
Starting program: /home/pi/orb/ofiles/orbuculum -f trace.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x76e12460 (LWP 4082)]
[New Thread 0x76611460 (LWP 4083)]

Thread 1 "orbuculum" received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/arm/armv6t2/strlen.S:126
126	../sysdeps/arm/armv6t2/strlen.S: No such file or directory.
(gdb) bt
#0  strlen () at ../sysdeps/arm/armv6t2/strlen.S:126
#1  0x76e9f344 in __vfprintf_internal (s=s@entry=0x7effe290, format=format@entry=0x406920 "%d,%ld,%d,%s,0x%x\n", ap=..., ap@entry=..., 
    mode_flags=mode_flags@entry=0) at vfprintf-internal.c:1688
#2  0x76eab0a4 in __vsnprintf_internal (string=0x7effe3dc "4,56,0,0", maxlen=<optimized out>, format=0x406920 "%d,%ld,%d,%s,0x%x\n", 
    format@entry=0x7effe3bc "\004", args=..., args@entry=..., mode_flags=mode_flags@entry=0) at vsnprintf.c:114
#3  0x76e92d26 in __GI___snprintf (s=<optimized out>, maxlen=<optimized out>, format=0x406920 "%d,%ld,%d,%s,0x%x\n") at snprintf.c:31
#4  0x004040aa in _handleDataRWWP ()
#5  0x0040446a in _itmPumpProcess ()
#6  0x00401f82 in _processBlock ()
#7  0x00402f5c in fileFeeder ()
#8  0x00401804 in main ()
(gdb) 
```

I actually got a lot of string format warnings when building, I thought that would be fine but now that I see this error it may actually have something to do with it?

If you don't have an immediate clue I will try with ubuntu instead of raspbian tomorrow. Did you use 20.04?

mubes, 2020-11-03:
I used 20.10 cos it's new and shiny. You'll get a depreciation warning but
I've already fixed that for the next commit.

Regards

Dave

mubes, 2020-11-03:
The dwt_segfault_fix contains a patch for the deprecation warning. It's also where any eventual patch for this segfault would end up going.

DAVE

mubes, 2020-11-03:
> Src/fifos.c: In function ‘_handleException’:
> Src/fifos.c:309:66: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 5 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Wformat=]
>   309 |         opLen = snprintf( outputString, MAX_STRING_LENGTH, "%d,%ld,%s,%s" EOL, HWEVENT_EXCEPTION, eventdifftS, exEvent[m->eventType & 0x03], exNames[m->exceptionNumber & 0x0F] );
>       |                                                                ~~^                                ~~~~~~~~~~~
>       |                                                                  |                                |
>       |                                                                  long int                         uint64_t {aka long long unsigned int}
>       |                                                                %lld

Now, this could easily cause a problem. AFAIK rasbian is 32 bit native so then all of the other parameters on that command line would be out by 32 bits.  Can you try changing that parameter to %lld and let's see if that fixes it? If it does then I guess we need to use the exact length format specifiers for printf, which we can easily do...but let's see if that's the problem first.

mubes, 2020-11-03:
Benjoyo,

It's flying a bit of a kite, but please try the current dwt_segfault_fix branch. It includes explicit 64 bit uint codings. I don't know if it will work but it's worth a fiddle with.

DAVE

Benjoyo, 2020-11-04:
> It's flying a bit of a kite, but please try the current dwt_segfault_fix branch. It includes explicit 64 bit uint codings. I don't know if it will work but it's worth a fiddle with.

Yes! That does work! The warnings for fifos.c are gone and it writes to hwevent file just fine, no segfault.

If it helps, this is the build output with the remaining warnings:

```
Compiling Src/generics.c
 Compiling Src/itmDecoder.c
 Compiling Src/tpiuDecoder.c
 Compiling Src/msgDecoder.c
 Compiling Src/msgSeq.c
Completed build of orb
 Compiling Src/orbuculum.c
 Compiling Src/filewriter.c
 Compiling Src/external/ftdispi.c
 Compiling Src/fifos.c
 Compiling Src/nwclient.c
Completed build of orbuculum
 Compiling Src/orbcat.c
Completed build of orbcat
 Compiling Src/orbtop.c
In file included from Src/orbtop.c:59:
Src/orbtop.c: In function ‘_outputTop’:
Inc/generics.h:67:19: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 5 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Wformat=]
   67 | #define C_YELLOW  "\033[1;33m"
      |                   ^~~~~~~~~~~~
Src/orbtop.c:637:38: note: in expansion of macro ‘C_YELLOW’
  637 |                     fprintf( stdout, C_YELLOW "%3d.%02d%% " C_LBLUE " %8ld ", percentage / 100, percentage % 100, report[n].count );
      |                                      ^~~~~~~~
Src/orbtop.c:637:74: note: format string is defined here
  637 |                     fprintf( stdout, C_YELLOW "%3d.%02d%% " C_LBLUE " %8ld ", percentage / 100, percentage % 100, report[n].count );
      |                                                                       ~~~^
      |                                                                          |
      |                                                                          long int
      |                                                                       %8lld
In file included from Src/orbtop.c:59:
Inc/generics.h:67:19: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 5 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Wformat=]
   67 | #define C_YELLOW  "\033[1;33m"
      |                   ^~~~~~~~~~~~
Src/orbtop.c:693:22: note: in expansion of macro ‘C_YELLOW’
  693 |     fprintf( stdout, C_YELLOW "%3d.%02d%% " C_LBLUE " %8ld " C_RESET "of "C_YELLOW" %ld "C_RESET" Samples" EOL, totPercent / 100, totPercent % 100, dispSamples, samples );
      |                      ^~~~~~~~
Src/orbtop.c:693:58: note: format string is defined here
  693 |     fprintf( stdout, C_YELLOW "%3d.%02d%% " C_LBLUE " %8ld " C_RESET "of "C_YELLOW" %ld "C_RESET" Samples" EOL, totPercent / 100, totPercent % 100, dispSamples, samples );
      |                                                       ~~~^
      |                                                          |
      |                                                          long int
      |                                                       %8lld
In file included from Src/orbtop.c:59:
Inc/generics.h:67:19: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 6 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Wformat=]
   67 | #define C_YELLOW  "\033[1;33m"
      |                   ^~~~~~~~~~~~
Src/orbtop.c:693:22: note: in expansion of macro ‘C_YELLOW’
  693 |     fprintf( stdout, C_YELLOW "%3d.%02d%% " C_LBLUE " %8ld " C_RESET "of "C_YELLOW" %ld "C_RESET" Samples" EOL, totPercent / 100, totPercent % 100, dispSamples, samples );
      |                      ^~~~~~~~
Src/orbtop.c:693:87: note: format string is defined here
  693 |     fprintf( stdout, C_YELLOW "%3d.%02d%% " C_LBLUE " %8ld " C_RESET "of "C_YELLOW" %ld "C_RESET" Samples" EOL, totPercent / 100, totPercent % 100, dispSamples, samples );
      |                                                                                     ~~^
      |                                                                                       |
      |                                                                                       long int
      |                                                                                     %lld
 Compiling Src/symbols.c
 Compiling Src/external/cJSON.c
Completed build of orbtop
 Compiling Src/orbdump.c
Completed build of orbdump
 Compiling Src/orbstat.c
Src/orbstat.c: In function ‘_dumpProfile’:
Src/orbstat.c:367:65: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 9 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Wformat=]
[... 3 lines cut ...]
      |                                                                 long int                                                                          uint64_t {aka long long unsigned int}
      |                                                               %lld
Src/orbstat.c:396:65: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 9 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Wformat=]
  396 |             fprintf( _r.c, "fl=(%d) %s\nfn=(%d) %s\n0x%08x %d %ld\n", t->index, t->n->filename, t->index, t->n->function, t->n->addr, t->n->line, myCost );
      |                                                               ~~^                                                                                 ~~~~~~
      |                                                                 |                                                                                 |
[... cut ...]
  475 |     fprintf( _r.c, "summary: %ld\n", _r.calls[_r.cdCount - 1].tstamp - _r.calls[0].tstamp );
      |                              ~~^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                |                                     |
      |                                long int                              uint64_t {aka long long unsigned int}
      |                              %lld
Completed build of orbstat
```

But for my current use case that fix was enough, thanks so much!

mubes, 2020-11-04:
OK, please try the latest version in dwt_segfault_fix, hopefully that 
removes the last of the warnings and we can fold this back into main.

Benjoyo, 2020-11-04:
Yes, all warnings are gone 👍

mubes, 2020-11-05:
OK, this is now folded into main, as at 9f6a8e9ad68bd0594e851c68dc9f98631358c2d2

closed 2020-11-05 by mubes