orbcode/orbuculum #46: Build error in Xubuntu 18.04 : `cannot find -liberty`

pvonmoradi, 2020-12-16:
I cloned the repo and tried building the main branch which resulted in this:
```
make WITH_FPGA=0 
 Compiling Src/generics.c
 Compiling Src/itmDecoder.c
 Compiling Src/tpiuDecoder.c
 Compiling Src/msgDecoder.c
 Compiling Src/msgSeq.c
Completed build of orb
 Compiling Src/orbuculum.c
 Compiling Src/filewriter.c
 Compiling Src/fifos.c
 Compiling Src/nwclient.c
/usr/bin/ld: cannot find -liberty
collect2: error: ld returned 1 exit status
Makefile:218: recipe for target 'orbuculum' failed
make: *** [orbuculum] Error 1
```
I was able to fix it (and meet other deps except FPGA) with : 
```
sudo apt install binutils-dev libelf-dev libiberty-dev
```
Thanks for this software :)

mubes, 2020-12-21:
Pooya, I'll keep this one open until I've made the documentation changes, otherwise I'll forget :-)
DAVE

mubes, 2021-01-24:
This is now fixed in https://github.com/orbcode/orbuculum/commit/c2be1b3a6258d8608172adc4438e911eaeeff2e2 so I'm closing this one. Thanks.

closed 2021-01-24 by mubes