orbcode/orbuculum #58: Discussion: RTIC Scope and orbuculum

tmplt, 2022-01-05:
Hello. This is a notice of another project which aims to accomplish similar goals as orbuculum. My aim here is to establish a line of communication that may eventually lead to some cross-pollination that would hopefully improve both projects.

---

As part of my thesis which I'm currently wrapping up I have been working with ITM on Cortex-M platforms where [RTIC](https://rtic.rs) is used. This has collimated into a toolset named [RTIC Scope](https://github.com/rtic-scope). Unfortunately, I did not discover orbuculum until later later into the thesis and only just recently realized that I have effectively reimplemented some features which orbuculum already offers. While orbuculum seems to be a more generally applicable UNIX-style toolset, RTIC Scope instead exclusively fits into the Rust ecosystem by offering a cargo subcommand which aims to do it all:
1. build your firmware with cargo;
2. recover some metadata;
3. optionally flash your target with the firmware and reset it;
4. collect trace data from the target using SWO or the embedded trace buffer;
5. map trace information to RTIC tasks (i.e. resolve raw ITM data);
6. save trace packets to file for port-mortem analysis; and
7. forward resolved trace information to frontends which can do whatever they like (display something in a GUI, save traces in a database, etc.).

The whole point of the toolset is to remove any end-user overhead: if an application is written in RTIC, RTIC Scope should allow instant insight into the system without any setup. We're some ways away from zero end-user overhead in the project's current state, however.

Interface-wise RTIC Scope currently supports any targets [`probe-rs`](https://probe.rs/) supports and serial devices on which a raw ITM data-stream is expected.

Aside from the cargo subcommand work has been made on APIs for configuring ITM tracing target-side and host-side (via `probe-rs`), and an [ITM protocol decoding library](https://github.com/rtic-scope/itm) has been written.

The next big feature for RTIC Scope would be a graphical display of the trace data which orbuculum would appear to offer via `orbtop` and `orbstat`. I'll be taking a look at these in the coming weeks and see if they can be combined with RTIC Scope.

---

Now that we're aware of eachother I look forward to any eventual cross-pollination between the two projects.

Cheers.

mubes, 2022-01-05:
Hi Viktor,

There's little doubt that Rust is the future, and the entire approach 
the community is taking to Debug is much more sensible than the one that 
C community did (arguably it learnt from the mistakes of its 
predecessor). A Rust based system should be based on Rust and Cargo 
etc...some folks have debugged Rust systems using Orbuculum, but I don't 
think that's a scalable solution to go forward with.

The only thing I would say is to try and avoid it being RTIC-specific. 
RTIM/RTIC is a pretty good framework but folks will want to use other 
things besides it...and indeed, it might form the starting point for a 
replacement for the existing C-based Orb utilities when it reaches 
maturity if it doesn't mandate RTIC at its core.

You might want to consider parallel trace as well as SWO, in which case 
checkout ORBTrace.

Good luck!

DAVE

tmplt, 2022-01-19:
Thanks for the tips!

mubes, 2022-01-22:
Viktor,

I'm going to close this just to keep things tidy, but please keep in contact and good luck with the thesis!

DAVE

closed 2022-01-22 by mubes