orbcode/orbuculum #69: orbtop: Exception number format, 0..N vs -16..N

Novakov, 2022-07-02:
`orbtop` is using numbers starting from 0 as exception number, it is also numbering scheme used in ARM docs. However typical definition of `IRQn_Type` enum will use negative numbers for exceptions and non-negative for interrupts (`SysTick_IRQn  = -1` and `FirstInterrupt = 0`). 

While translating between these two schemes are trivial it is also easy for human being to make off-by-one error and mismatch interrupts (I imagine that two interrupts in active use can be next to each other so mistake will not be easy to spot).

I'm not sure that switching `orbtop` numbering to -16..N is the best choice but maybe adding some info ("subtract 16 to get IRQn_Type values") or configurable option?

zyp, 2022-07-02:
I'd argue that when only displaying a number, it makes most sense to use the same range as the IPSR register for consistency, i.e. exception numbers with IRQ0 showing as exception 16. Having to add 16 to a negative number to find the exception number seems to me more inconvenient and error prone than having to subtract 16 to find the IRQ number from an exception number.

In longer form, it makes good sense to split the ranges and distinguish between core exceptions and IRQs, or even display both; e.g: `Exception 3 (HardFault)` and `Exception 18 (IRQ 2)`.

Novakov, 2022-07-02:
I would say that depends in what context you are looking, I see arguments for for schemes.

I really like the idea of displaying both - exception number with name and IRQ number. It would be even better to supply IRQ names from file and display them as well but that's just my wishful thinking ;)

mubes, 2022-07-02:
Well, it's trivial to add a command line option to emit negative numbers if
you want them. I have to say I do like the idea of 'named' exceptions and
interrupts. We can obviously do that for the standardised ones, but will
need SVDs to do that meaningfully, which is another tedious configuration
requirement.

It's going to be a low priority thing though, unless someone wants to
contribute a patch...

Dave

Novakov, 2022-07-03:
https://github.com/orbcode/orbuculum/pull/70 - only standard exception names, no SVD parsing

mubes, 2022-07-03:
Hey, waddya know, a patch. Thanks!  This is fixed via #3db5900.

closed 2022-07-03 by mubes