orbcode/orbuculum #99: Orbuculum fails to connect to J-LINK GDB Server on Windows.

markuckermann, 2023-03-09:
OS: Windows 10
J-LINK: On board of an nrf52840DK
`jlinkgdbservercl -select USB -device nRF52840_xxAA -endian little -if SWD -speed 4000 -noir
 -LocalhostOnly -nologtofile`
 
GDB:
```
target remote localhost:2331
monitor reset
monitor SWO EnableTarget 0 8000000 0xff 0
prepareSWO 64000000 8000000
... other standard setups to enable PC sampling ...
```
`orbuculum -s localhost:2332 -m 1000 -v3` outputs
```
orbuculum version a4d410b-dirty
Report Intv    : 1000 mS
NW SERVER H&P  : localhost:2332
Use/Strip TPIU : False
Established NW Server Link
Lost NW Server Link
Established NW Server Link
 No active connection
 ```
 pretty much immediatly. 
 
 However, orbtop can connect fine using `orbtop -s localhost:2332`
 
 Let me know if there is anything I can help with. Love the project ❤️

Novakov, 2023-05-26:
I can confirm the same behaviour with J-Link GDB Server + EFM32GG on Windows 10. In my case `orbcat` is outputing when connected directly connected to J-Link GDB Server but in `orbuculum` fails to connect. I will need few days to investigate further.

Novakov, 2023-06-04:
@markuckermann Could you try https://github.com/orbcode/orbuculum/pull/103 (artifacts here: https://github.com/orbcode/orbuculum/actions/runs/5170474165)?

mubes, 2023-06-21:
Closing this as fixed by https://github.com/orbcode/orbuculum/commit/cf50f95f898996ff43622a82f0c7c71c4e75ba15 
Please re-open if you have further issues.

closed 2023-06-21 by mubes