orbcode/orbuculum #102: Inconsistent results from orbfifo when reading from file and writing to permanent file

jpcrypt, 2023-04-03:
I was previously using an older version of orbuculum with this incantation (input trace file: [raw.zip](https://github.com/orbcode/orbuculum/files/11140999/raw.zip)):
`orbuculum -t -f raw.bin -P -e`

and this worked very well without any issues. The expected output (obtained from the 1.13 release) is:
```
8,21,0x00003bc0
8,21,0x00003aa8
8,21,0x00003bc0
8,21,0x00003aa8
8,21,0x00003bc0
8,21,0x00003aa8
8,21,0x00003bc0
8,21,0x00003aa8
8,21,0x00003bc0
8,21,0x00003aa8
8,21,0x00003bc0
8,21,0x00003aa8
8,21,0x00003bc0
8,21,0x00003aa8
8,21,0x00003bc0
8,21,0x00003aa8
8,21,0x00003bc0
```
(this is the contents of the generated hwevents file)

Now I'm trying to do the same from the latest code (git main branch) and the results are inconsistent. Sometimes I get the same, sometimes I get a truncated version with fewer lines. My command line is:
`orbfifo -t 1 -E -P  --input-file raw.bin -c 0,blah,'%08x\n' -v 3`

I do seem to always get the full output if I omit the -E option.
Another issue is that `hwevent` is either not generated at all, or empty, if I omit the `-v 3` and/or `-c 0...` options. (I do not care about the 'blah' output file; hwevents is all I'm after.)

I poked around a bit and I suspect the truncation problem may lie here: https://github.com/orbcode/orbuculum/blob/a4d410b4bdaffc286fc4ac7281436905d0258e27/Src/orbfifo.c#L448
because I move the `_processBlock()` prior to the `RESULT_EOF / RESULT_ERROR` check, then the output appears to always be complete. However trying to debug and observe what's happening when the output gets truncated seems to make the problem go away (maybe some race condition?).

mubes, 2023-05-21:
Hiya,
Could you please give c106e13513bf45e72e085144905302969f37459e a try to see if that helps with your problem? I think the issue was that we simply weren't cleanly closing the fifos for the particular configuration you were using and that lead to a race.  Please let me know how you get on.
DAVE

mubes, 2023-05-21:
OK, second bite of the cherry, was a bit of silliness with the termination handling. Please try 3a5d82b and re-open this if it still gives you grief. At least in my testing that is all working nicely for the case you sent;
```
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ rm hwevent; orbfifo -t 1 -E -P  --input-file ~/Desktop/raw.bin ; wc hwevent 
 17  17 272 hwevent
$ 
```

closed 2023-05-21 by mubes

jpcrypt, 2023-05-23:
looks good, thank you!