orbcode/orbuculum #105: Arch linux builds need a meson prefix

Scottapotamas, 2023-07-06:
Got my orbtrace mini today and I've been speedrunning getting all the tooling going, coming from a SeggerGDBServer style setup.

## Issue

Blindly following this repo's docs will lead overly keen Arch Linux users into a pretty typical error when any orb\* tool tries to load liborb.so from the default install path `/usr/local/lib/liborb.so`.

```
orbtrace: error while loading shared libraries: liborb.so.0: cannot open shared object file: No such file or directory
```

## Resolution

[Arch Linux's packaging etiquette](https://wiki.archlinux.org/title/Arch_package_guidelines#Package_etiquette) normally wants meson builds to be prefixed on `/usr`.

i.e. a more correct incantation for this repo's build is:

```
meson setup --prefix=/usr build
```

___

I'm happy to PR a note into the docs if you want.

mubes, 2023-07-10:
Hiya,
Yes, please generate a PR and add yourself to the CONTRIBUTORS file too. I can obviously make the change but it's easy to make a typo or miss something that's essential for the Arch case, whereas you can cut/paste it directly from a session.
Thanks
DAVE

zyp, 2023-07-10:
I don't think this is an Arch issue, all distros tends to want *packages* installed under the `/usr` prefix. The issue here is that you're not building a package, you're building nonpackaged software, and the convention is to install that under `/usr/local` to keep it separate from packaged software.

That said, the best way to solve this for Arch would be to write a PKGBUILD and have it included in AUR, so it becomes a package that gets installed in `/usr`. 🙂

mubes, 2023-07-16:
Fixed with a note in the readme in #e3dba63. Thanks for the info.

closed 2023-07-16 by mubes