orbcode/orbuculum #116: Cannot abort orbfifo with Ctrl + C

LnnrtS, 2023-08-02:
It just does not react.  
Orbcat for example works fine

mubes, 2023-08-02:
Thanks for the report. This should be resolved in https://github.com/orbcode/orbuculum/commit/a48d03ac08948093a5712172d6bef83a3d2ec024 . Could you please take a look and provide feedback.

LnnrtS, 2023-08-03:
Works now, thanks!

LnnrtS, 2023-08-03:
No I cannot abort while connected

mubes, 2023-08-03:
Well, that wasn't good, was it?  Let's try again with https://github.com/orbcode/orbuculum/commit/1ca6e07ff2371e74eee9b8b2f3ca97a63b4d9306 . Sorry about that.  Once we've got this one cleared away could you please take a quick look at the other two and close those off it they're resolved too please?

mubes, 2023-08-04:
I did a bit more testing with this. Everything was working but CPU utilisation was very high. I've improved that. As far as I know https://github.com/orbcode/orbuculum/commit/4782fdce74acadfabfc67c6a599591f17dc061a2 should resolve all of your issues...let me know how you get on and then I'll merge it into main.

mubes, 2023-08-07:
I have merged these fixes into main since they are not controversial and are of benefit to all users. If you have any more problems please re-open this.

closed 2023-08-07 by mubes