orbcode/orbuculum #120: gdbtrace.init does not seem to work with newer version of arm-none-eabi-gdb

ClemensElflein, 2023-08-30:
I have tried using orbuculum with the toolchain linked here: https://modm.io/guide/installation/

The issue is that the arm-none-eabi-gdb (version 12.1.90) does not correctly process the `gdbtrace.init` from this repository and therefore the enableSTM32TRACE command cannot be used.

Example output is here: https://cdn.discordapp.com/attachments/614885210395508738/1146444609174654987/image.png

salkinium, 2023-08-31:
Weird, which platform did you try this on? Windows perhaps?
I use both the ARM (non-Python) and the xpack (Python3) GDB v12 on macOS and it works fine.

ClemensElflein, 2023-08-31:
PopOS 22.04, so basically Ubuntu 22.04

niklaut, 2023-09-22:
This exact problem is now also happening with my GDB after I updated the `gdbtrace.init` to the latest version.

zyp, 2023-09-22:
Which platform are you on? I haven't been able to reproduce it yet.

niklaut, 2023-09-22:
I'm on macOS 13.5.2 with xpack arm-none-eabi-gdb-py3
```
GNU gdb (xPack GNU Arm Embedded GCC arm64) 12.1.90.20221210-git
```

It's weird, cos parts of the script are shown with autocompletion:

```
(gdb) _s
_setAddressesEFR32MG12  _setAddressesETM4       _setAddressesIMXRT      _setAddressesNRF        _setAddressesNRF52      _setAddressesNRF53      _setAddressesSTM32      _setAddressesTM4C       _startETMv35            _startETMv4
(gdb) start
start     startETM  starti
(gdb) stop
stop     stopETM
(gdb) describeETM
(gdb) prepa
```
but then it stops after `describeETM`, so something in it breaks it. (there's a duplicate stopETM function also btw)
if I remove the `describeETM` function, I can call the `enableSTM32TRACE` function.

I can see that it configures the pins, but the orbtrace's trace led doesn't turn red. I suspect that this is a hardware issue with my wiring though.

zyp, 2023-09-22:
Thanks, reproduced and fixed in 7809380. `describeETM` was missing an `end` at the end, so it ate the rest of the script, and the way it was indented didn't make it obvious.

closed 2023-09-22 by zyp

niklaut, 2023-09-25:
Yup, that fixed it for me, thanks!

commit 7809380f (2023-09-22): gdbtrace.init: Fix describeETM definition.

Fixes #120