orbcode/orbuculum #123: Windows CLI output has corrupted characters

smatt123, 2023-09-21:
Hi,

Thanks for providing this great tool!

When using the provided win64 releases (tested v2.1.0), some part of the outputs of the tools appear to be corrupted. Maybe this is an issue with wrong character encodings?

Examples: 
`orbuculum -m 1000 -s localhost:50001
←[1F←[K←[1;33m     0 ←[0m Bits/sec ←[0m
←[1F←[K←[1;33m     0 ←[0m Bits/sec ←[0m
←[1F←[K←[1;33m     0 ←[0m Bits/sec ←[0m
←[1F←[K←[1;33m     0 ←[0m Bits/sec ←[0m`

`orbtop -e test.elf
←[1;33mLoaded test.elf
←[0m←[2J←[;HConnected...
←[2J←[;H←[1;33m 61.57% ←[1;34m    1849 ←[1;36m** Sleeping **←[0m
←[1;33m 33.99% ←[1;34m    1021 ←[1;36mvApplicationIdleHook←[0m
←[1;33m  2.33% ←[1;34m      70 ←[1;36mFLEXCOMM1_DriverIRQHandler←[0m
←[1;33m  0.79% ←[1;34m      24 ←[1;36mprvIdleTask←[0m
←[1;33m  0.46% ←[1;34m      14 ←[1;36mFLEXCOMM_GetInstance←[0m
←[1;33m  0.19% ←[1;34m       6 ←[1;36mCTIMER_GetStatusFlags←[0m
←[1;33m  0.13% ←[1;34m       4 ←[1;36mSPI_MasterTransferNonBlocking←[0m
←[1;33m  0.13% ←[1;34m       4 ←[1;36mprvCheckTasksWaitingTermination←[0m
←[0m-----------------
←[1;33m 99.59% ←[1;34m    2992 ←[0mof ←[1;33m 3003 ←[0m Samples

←[0m[←[1;31mV←[1;32mS←[0m-←[1;36mH←[0m] ←[0mInterval = ←[1;33m998←[0mms`

Thanks!

Novakov, 2023-09-21:
Hi

The problem is not with orbuculum itself but with shell and terminal emulator that does not interpret correctly ANSI color codes. 

What shell/terminal are you using? I will take a look if some something can be done to detect problematic shells.

On Windows I recommend using Windows Terminal + pwsh

smatt123, 2023-09-21:
Thanks for the quick response! I can confirm that it works well in Windows Terminal :)

closed 2023-09-21 by smatt123