orbcode/orbuculum #141: Orbmortem save to file not working

augustofg, 2024-02-29:
We are trying to save the captured instruction trace with `orbmortem`. The help menu says that you should use the `S` key for this, but after typing the filename and hitting 'enter' the file isn't created anywhere. I would expect for the file to be saved in the current working directory, but I see nothing.

Am I missing something?

Thanks,
Augusto.

mubes, 2024-03-11:
In my testing this is fixed via https://github.com/orbcode/orbuculum/commit/ab66c1bf548808a8cad7e125ac3adf38f1467c75 ... please give it a try and see how you get on.

Thanks for the report, sorry it took a few days to get to it.

DAVE

augustofg, 2024-03-12:
Thanks Dave, it works perfectly now!

closed 2024-03-12 by augustofg