orbcode/orbuculum #154: orbtop works, orbstat doesn't

maxgerhardt, 2024-08-28:
I use a Nucleo-L476RG (STM32L76RG, Cortex-M4). I use libtrace to activate SWO UART output at 2Mhz (80MHz CPU speed), periodic PC sampling every 64 clock cycles, TPIU formatter off etc in the firwmare itself. 

I upload the firmware (does blinky + enable trace), then start OpenOCD:

>C:\Users\Max\.platformio\packages\tool-openocd\bin\openocd -d2 -f interface/stlink.cfg -f target/stm32l4x.cfg -c "tpiu config internal :3443 uart off 80000000 2000000"

Then I run `orbtop`, and it works corectly:

>C:\Users\Max\Downloads\orbuculum-win64\bin\orbtop.exe -e "C:\Users\Max\temp\l4_trace\.pio\build\nucleo_l476rg\firmware.elf"  -n -o live.txt -r routines.txt -g record.txt -E  -I 2000

[image]

This looks sensible for a delay(1000) + GPIO write.

Now I want to use `orbstat` to analyze the trace later in KCachegrind. I cannot find any combinations of options in which this tool works though.

```
orbstat  -e "C:\Users\Max\temp\l4_trace\.pio\build\nucleo_l476rg\firmware.elf"  -v3
orbstat version 233c34e-dirty
Server          : localhost:3443
Delete Material : None
Elf File        : C:\Users\Max\temp\l4_trace\.pio\build\nucleo_l476rg\firmware.elf (Don't Truncate)
DOT file        : None
ForceSync       : true
Trace Channel   : 30
Sample Duration : 1000 mS
Objdump options  : None

arm-none-eabi-objdump -Sl -C --source-comment=sRc## C:\Users\Max\temp\l4_trace\.pio\build\nucleo_l476rg\firmware.elf

Loaded C:\Users\Max\temp\l4_trace\.pio\build\nucleo_l476rg\firmware.elf
ITM Overflow (1)
ITM Overflow (2)
ITM Overflow (3)
ITM Overflow (4)
ITM Overflow (5)
..

```
If I tell it to enable the TPIU formatter on channel 1, it does not throw any overflow packets, but it also detects 0 changes when I Ctrl+C.

```
arm-none-eabi-objdump -Sl -C --source-comment=sRc## C:\Users\Max\temp\l4_trace\.pio\build\nucleo_l476rg\firmware.elf

Loaded C:\Users\Max\temp\l4_trace\.pio\build\nucleo_l476rg\firmware.elf
Received 3996682 raw sample bytes, 0 function changes, 0 distinct addresses
```

Any hints on how to get this working?

mubes, 2024-08-28:
> clock cycles, TPIU formatter off etc in the firmware itself. You can find
> the code at https://github.com/maxgerhardt/stm32l4_swo_trace.

Hi,

Your speed looks very low. The ITM Overflows imply that you are sending
additional information from the hardware side of the ITM over the channel
... perhaps you've got timestamps switched on? I don't know how fast
openocd can send data and/or if any get lost in the process I'm afraid, but
it looks as if the channel is just over-full.  I'd suggest removing the
timestamps and/or speeding up the SWO if your probe can do that? Turning
off the PC samples while performing stat collection will also help -
basically, you've got one channel and it's relatively small in your config,
so we need to make sure it's used optimally.  ORBTrace can collect SWO at
about 6MBytes/sec but I don't know what probe you are using (or indeed what
speed openocd can reliably deliver).

Note that orbstat is probably a bit rusty - I don't think anyone has used
it in a while, but I am very happy to fix up any problems you might find.
First of all though we need to get that SWO channel reliable...if messages
are missing then it's very unlikely the flow can be reconstructed reliably.
Are you following the notes from
https://orbcode.org/orbuculum/swo-more-instrumentation/ ?

Regards

DAVE

maxgerhardt, 2024-08-28:
>Turning off the PC samples while performing stat collection will also help 

I may have a fundamental misunderstanding here, but is the periodic PC sampling data output over SWO not exactly what `orbstat` *captures*? It sends its own commands via SWO to facilitate the tracing? (Not I only have SWO, no TRACEDATA pins available).

(And, the ITM overflows seem to have been resolved by turning off the ITM EnableSyncPacket, local and global timestamps were already off).

I'm using the on-board ST-Link V2 of the Nucleo-L476RG (max SWO frequency of 2MHz) for this. My idea was that I can get an (approximat, every 64 clock cycles or more) execution trace with this periodic PC sampling into that kcachegrind to view graphically.

mubes, 2024-08-28:
The sampling from the PC interval reporting is only a snapshot of what the
chip executing, it isn't sufficient to rebuild a trace of execution, but
it's perfect for figuring out where time is being spent.

Orbstat works by adding instrumentation to function entry and exits and
then reporting those over the itm channel...you will need to have
instrumented your code for that to report anything, otherwise you will get
zero samples...there simply isn't anything to report.

There are other utilities are are aiming closer for what I think you want
to do. Orbprofile uses explicit instruction tracing over the ETM channel to
reconstruct an execution record but (a) that needs parallel trace hardware
and (b) it is _very_ experimental, its a long way from being usable yet.

Note that you can report function entry and exit events using any suitable
Comms medium, it doesn't need to be ITM, so you could do that using a UART
link, if that was more convenient, for example.

For interactive support please feel free to join the discord
channel...there are usually folks around who can help out directly.

Dave

maxgerhardt, 2024-08-29:
I understand now how `orbstat` is supposed to be used in the context of compiler instrumentalization. Per https://orbcode.org/orbuculum/swo-code-instrumentation/ and libtrace it worked on my Nucleo-L476RG and a Nucleo-WL55JC1. `orbstat` reported valid function changes and output it to the KCacheGrind and the dotfile file. While the output in KCacheGrind is not exactly as I expected and some numbers look funky (switching to"relative" mode often gives over 50000% in the "Inst." column), I'll work on some tooling myself to get an [execution timeline](https://github.com/gaogaotiantian/viztracer/blob/master/img/example.png)

mubes, 2024-08-29:
No worries. Feel free to polish the stuff we've got if that floats your
boat. Contributions always welcome :-)