orbcode/orbuculum #164: Can't configure GDB session for STM32H7

diggit, 2025-03-30:
Hi,
I have a custom STM32H730 and J-Link. My goal is to get `orbtop` working. right now, I am stuck at GDB throwing error in case of H7 (11)

GDB stops with:
```
SWO enabled successfully.
.gdbinit:17: Error in sourced command file:
Invalid type combination in equality test.
```
There is something wrong in `enableSTM32SWO` or other "sequences"/macros. unfortunately GDB does not say where.

If I change the call to `enableSTM32SWO 1` , I don't get this error.

I think it will be something trivial, but gdb not telling where the error is, does not help.

my `.gdbinit` is:
```sh
set verbose on 
source /path/to/Support/gdbtrace.init 
target extended-remote localhost:50000
#monitor swdp_scan
file build/binary.elf
#attach 1
set mem inaccessible-by-default off
set print pretty
load

start

monitor SWO EnableTarget 500000000 2000000 0xFF 0

# ----------Stm32H7 as debuggee----------------
enableSTM32SWO 11

prepareSWO SystemCoreClock 500000000 1 0

dwtSamplePC 1
dwtSyncTap 3
dwtPostTap 1
dwtPostInit 1
dwtPostReset 15
dwtCycEna 1

ITMId 1
ITMGTSFreq 3
ITMTSPrescale 3
ITMTXEna 1
ITMSYNCEna 1
ITMEna 1

ITMTER 0 0xFFFFFFFF # Enable Trace Ports
ITMTPR 0xFFFFFFFF # Make them accessible to user level code
```

mubes, 2025-03-30:
Try enableSTM32SWO 7 since this is a H7 device. In general `help <command>`
should give you a bit of info.

Regards

Dave

jmalmari, 2025-04-02:
The 7 is actually just F7, see table here: https://github.com/orbcode/orbuculum/pull/162#issue-2858661481

You might want to check your DBGMCU ID to see if autodetection works.

closed 2025-04-09 by diggit